3dtof/voxelsdk #162: make -j2  errors

xubin4952, 2018-08-07:
hi，
when I execute make -j2 in build I got the errors below ，and I have installed the swig3.0.
does anyone know how to fix this problem ?  thank you!

[ 61%] Building CXX object VoxelPCL/CMakeFiles/voxelpcl.dir/PCLGrabber.cpp.o
/home/jason/TI-tof/libvoxel/voxelsdk/Voxel/SWIG/../Device.h:63: Error: Syntax error in input(3).
make[2]: *** [lib/python2.7/VoxelPYTHON_wrap.cxx] Error 1
make[1]: *** [Voxel/SWIG/CMakeFiles/_Voxel.dir/all] Error 2
make[1]: *** Waiting for unfinished jobs....
Linking CXX shared library ../lib/libvoxelpcl.so

suramyag1, 2018-08-07:
Hi @xubin4952, 

We do not see this error.
However, a web search found [this](https://stackoverflow.com/questions/18213813/swig-syntax-error-in-input3). This says that the error is for C++99 standards and not C++11. 
Since it's pointing to line 63, you may want to put an extra space between the ">>". 
`
virtual Vector<Ptr<Device> > getDevices(const Vector<int> &channels) const { return Vector<Ptr<Device> >(); }
` 
Hope this helps

Suramya

suramyag1, 2018-08-08:
@xubin4952 

Please disable the "Generate Python Bindings" and build. You'll be able to use the CLI to capture the data. 

Suramya

xubin4952, 2018-08-08:
Hi,
I will have a try . Can you explain in detail how to operate？  I am not familiar for this. Thank you !

suramyag1, 2018-08-08:
For building without python bindings, look at https://github.com/3dtof/voxelsdk/wiki/Build-from-source-on-Linux#optional-pre-requisites

For running the CLI, refer https://github.com/3dtof/voxelsdk/wiki/Voxel-CLI

xubin4952, 2018-08-25:
Hi， 
thank  you ,now I can build the code ,I want to know  how to create tools used in linux system.I want to get depth data in my arm board.

suramyag1, 2018-09-04:
Please read the wiki for instructions on using VoxelSDK:
https://github.com/3dtof/voxelsdk/wiki/Voxel-SDK-User-Guide

VoxelCLI allows you to capture data, you can refer to the capture data from there to get phase.

Suramya