3dtof/voxelsdk #163: why does every Test projects in voxelsdk is error?and can not Debug

pans0ul, 2018-08-13:
Ive build voxlesdk Solution with instruction  :https://github.com/3dtof/voxelsdk/wiki/Build-from-source-on-Windows.

I could only run and debug VoxelCLI & simpleVoxel projects ,but canot do every Test projects.

error shows below:

[image]

[image]

`'CameraSystemReadStreamTest.exe' (Win32): Loaded 'C:\Users\ZC-WK-06\Desktop\voxelsdk-0.6.3\build\bin\CameraSystemReadStreamTest.exe'. Module was built without symbols.
'CameraSystemReadStreamTest.exe' (Win32): Loaded 'C:\Windows\System32\ntdll.dll'. Symbols loaded.
'CameraSystemReadStreamTest.exe' (Win32): Loaded 'C:\Windows\System32\kernel32.dll'. Symbols loaded.
'CameraSystemReadStreamTest.exe' (Win32): Loaded 'C:\Windows\System32\KernelBase.dll'. Symbols loaded.
'CameraSystemReadStreamTest.exe' (Win32): Loaded 'C:\Windows\System32\msvcr120.dll'. Symbols loaded.
'CameraSystemReadStreamTest.exe' (Win32): Loaded 'C:\Windows\System32\msvcp120.dll'. Symbols loaded.
'CameraSystemReadStreamTest.exe' (Win32): Loaded 'C:\Users\ZC-WK-06\Desktop\voxelsdk-0.6.3\build\lib\voxel.dll'. Module was built without symbols.
'CameraSystemReadStreamTest.exe' (Win32): Loaded 'C:\Windows\System32\setupapi.dll'. Symbols loaded.
'CameraSystemReadStreamTest.exe' (Win32): Loaded 'C:\Windows\System32\msvcrt.dll'. Symbols loaded.
'CameraSystemReadStreamTest.exe' (Win32): Loaded 'C:\Windows\System32\cfgmgr32.dll'. Symbols loaded.
'CameraSystemReadStreamTest.exe' (Win32): Loaded 'C:\Windows\System32\ucrtbase.dll'. Symbols loaded.
'CameraSystemReadStreamTest.exe' (Win32): Loaded 'C:\Windows\System32\rpcrt4.dll'. Symbols loaded.
'CameraSystemReadStreamTest.exe' (Win32): Loaded 'C:\Windows\System32\user32.dll'. Symbols loaded.
'CameraSystemReadStreamTest.exe' (Win32): Loaded 'C:\Windows\System32\win32u.dll'. Symbols loaded.
'CameraSystemReadStreamTest.exe' (Win32): Loaded 'C:\Windows\System32\gdi32.dll'. Symbols loaded.
'CameraSystemReadStreamTest.exe' (Win32): Loaded 'C:\Windows\System32\version.dll'. Symbols loaded.
'CameraSystemReadStreamTest.exe' (Win32): Loaded 'C:\Windows\System32\gdi32full.dll'. Symbols loaded.
'CameraSystemReadStreamTest.exe' (Win32): Loaded 'C:\Windows\System32\msvcp_win.dll'. Symbols loaded.
'CameraSystemReadStreamTest.exe' (Win32): Loaded 'C:\Windows\System32\ole32.dll'. Symbols loaded.
'CameraSystemReadStreamTest.exe' (Win32): Loaded 'C:\Windows\System32\combase.dll'. Symbols loaded.
'CameraSystemReadStreamTest.exe' (Win32): Loaded 'C:\Windows\System32\bcryptprimitives.dll'. Symbols loaded.
'CameraSystemReadStreamTest.exe' (Win32): Loaded 'C:\Windows\System32\sechost.dll'. Symbols loaded.
'CameraSystemReadStreamTest.exe' (Win32): Loaded 'C:\Windows\System32\oleaut32.dll'. Symbols loaded.
'CameraSystemReadStreamTest.exe' (Win32): Loaded 'C:\Windows\System32\vcomp120.dll'. Symbols loaded.
'CameraSystemReadStreamTest.exe' (Win32): Loaded 'C:\Windows\System32\imm32.dll'. Symbols loaded.
'CameraSystemReadStreamTest.exe' (Win32): Loaded 'C:\Windows\System32\dtrampo.dll'. Cannot find or open the PDB file.
'CameraSystemReadStreamTest.exe' (Win32): Loaded 'C:\Windows\System32\advapi32.dll'. Symbols loaded.
The thread 0x1de0 has exited with code -1 (0xffffffff).
The thread 0x3fe8 has exited with code -1 (0xffffffff).
The thread 0xbf0 has exited with code -1 (0xffffffff).
The thread 0x614 has exited with code -1 (0xffffffff).
The program '[14388] CameraSystemReadStreamTest.exe' has exited with code -1 (0xffffffff).
`

pans0ul, 2018-08-13:
Ive us Cmake to build solution ,didnt use visual studio to do it

pans0ul, 2018-08-16:
@alexfeinman @jayjevon @raylin @gadiyar

pans0ul, 2018-08-23:
T_T

suramyag1, 2018-09-04:
@pans0ul 
I hope you're using the correct arguments with the tests. Like CameraSystemReadStreamTest takes in the filename to dump frames. 

`
CameraSystemReadStreamTest -f test
`
should work fine. 

Suramya