3dtof/voxelsdk #169: Can I use it as a multi-camera?

ohsh16, 2019-02-25:
I want to use two or more cameras.
Can I run more than one camera on one PC?

ohsh16, 2019-02-26:
1)
FilterPtr CameraSystem :: createFilter (const String & name, DepthCamera :: FrameType type)

2)
bool CameraSystem :: disconnect (const DepthCameraPtr & depthCamera, bool reset)

Is it possible to perform multi-operation by entering DepthCameraPtr in the argument value of the createrFile function?

In the disconnect function, DepthCameraPtr argument value is included, and only DepthCameraPtr is used to disconnect the id value.

It is supposed that the setting value of DepthCameraPtr should be set in all setting functions so that it can be set at the time of multi-operation.

ohsh16, 2019-02-26:
bool DepthCamera :: start ()
How do you think the variables used in the function should be array, so multi-operation is possible?

suramyag1, 2019-03-05:
@ohsh16 
Filters are added to different depth cameras using the addFilter function from DepthCamera.cpp

You should probably create your own class for multiple cameras, to do each function together for multi-camera operation.