3dtof/voxelsdk #170: Build issues

leslierice1, 2019-03-05:
Hi, I followed all the instructions in the wiki for Installation on Windows and Build from source on Windows. I'm getting the following error when trying to import Voxel in a python script. 

`Traceback (most recent call last):
  File "CameraSystemTest.py", line 8, in <module>
    import Voxel
  File "C:\Program Files\Voxel SDK 0.6.10\lib\python2.7\Voxel.py", line 17, in <module>
    _Voxel = swig_import_helper()
  File "C:\Program Files\Voxel SDK 0.6.10\lib\python2.7\Voxel.py", line 16, in swig_import_helper
    return importlib.import_module('_Voxel')
  File "C:\Python27\lib\importlib\__init__.py", line 37, in import_module
    __import__(name)
ImportError: DLL load failed: The specified module could not be found.`

Additionally, I built the code for the DepthCapture example, and when I try and run the executable, I get a system error `The code execution cannot proceed because voxel.dll was not found. Reinstalling the program may fix this problem`. 

I did not get any errors when configuring, generating, and building using CMake, but have a feeling the build for voxelsdk was not successful.

suramyag1, 2019-03-05:
Hi @leslierice1, 

Are environment variables updated? You have to add the variables listed [here](https://github.com/3dtof/voxelsdk/wiki/Build-from-source-on-Windows#running-sdk-programs) to the path.

leslierice1, 2019-03-05:
I did that as well - what should the output of the build look like?

leslierice1, 2019-03-05:
Can I use visual studio 2017 or must I use visual studio 2013?

suramyag1, 2019-03-06:
If you use VS2017, you'll need the 2017 builds for Qt as well as PCL. 
If you don't want to build, you can directly use the 0.6.10 release, you just need to update the environment variables.

- VOXEL_SDK_PATH=<Installation directory>
- Add <Insallation directory>/bin;<installation directory>/bin to PATH variable
- Add <PCL path> and <Qt Path> to PATH variable

For VS2017, you'll need PCL 1.8.1: https://github.com/PointCloudLibrary/pcl/releases/download/pcl-1.8.1/PCL-1.8.1-AllInOne-msvc2017-win64.exe
Also, Qt with VS2017 release

leslierice1, 2019-03-14:
Thanks, I got the Python bindings working, but switched to using VS2013 and the versions of PCL and Qt recommended. Although when opening VoxelCLI I get System error: "The code execution cannot proceed because pcl_common_release.dll was not found".  

I was also wondering what the output format of the data when using the FRAME_RAW_FRAME_UNPROCESSED config (when using 1 subframe, 4 quad), I get an array of 8*320*240. How is this data ordered and what does it represent?

suramyag1, 2019-03-15:
@leslierice1 - For running VoxelCLI, add pcl dir/lib to your PATH as well. 

Raw unprocessed data is the data directly given from the sensor. Refer [OPT9221 datasheet](http://www.ti.com/lit/ds/symlink/opt9221.pdf), section 7.3.4. 

Please ask other questions related to OPT9221/OPT8241 on [e2e.ti.com](e2e.ti.com)

closed 2019-03-15 by suramyag1