3dtof/voxelsdk #171: prebuild VoxelSDK install on ubuntu16.04

robust-lu, 2019-03-07:
I follow the instructions of [https://github.com/3dtof/voxelsdk/wiki/Installation-on-Linux] to install VoxelSDK 0.6.10 on ubuntu16.04. When I run sudo dpkg -i libvoxel*.deb libti3dtof*.deb, it shows:

dpkg: dependency problems prevent configuration of libti3dtof-dev:
 libti3dtof-dev depends on libvoxel-dev (>= 0.6.10); however:
  Package libvoxel-dev is not installed.

but there is no libvoxel-dev package in 0.6.10, how to solve it?

suramyag1, 2019-03-07:
@robust-lu 

I've updated the release. It should work now.

robust-lu, 2019-03-08:
OK, it works.

closed 2019-03-08 by robust-lu

arenas7307979, 2019-12-19:
follow the instructions of [https://github.com/3dtof/voxelsdk/wiki/Installation-on-Linux] to install VoxelSDK 0.6.10 on **ubuntu18.04**. When I run sudo dpkg -i libvoxel*.deb libti3dtof*.deb, it shows:

dpkg: dependency problems prevent configuration of libti3dtof-dev:
libti3dtof-dev depends on libvoxel-dev (>= 0.6.10); however:
Package libvoxel-dev is not installed.

but there is no libvoxel-dev package in 0.6.10, how to solve it?